husarnet/husarnet-esp32-platformio #2: Eroor when building platforino

Grey-Lancaster, 2022-06-06:
PACKAGES:
 - framework-arduinoespressif32 @ 0.0.0
 - tool-esptoolpy @ 1.30300.0 (3.3.0)
 - tool-mkfatfs @ 2.0.1
 - tool-mklittlefs @ 1.203.210628 (2.3)
 - tool-mkspiffs @ 2.230.0 (2.30)
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch3
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 29 compatible libraries
Scanning dependencies...
Dependency Graph
|-- AsyncTCP @ 1.1.1+sha.e3b5f17
|-- Husarnet ESP32 @ 1.2.0-4
|-- ESP Async WebServer @ 1.2.3
|   |-- AsyncTCP @ 1.1.1+sha.e3b5f17
|   |-- FS @ 1.0
|   |-- WiFi @ 1.0
|-- Husarnet @ 1.0
|-- WiFi @ 1.0
Building in release mode
Compiling .pio\build\esp32dev\src\simple-webserver.cpp.o
Generating partitions .pio\build\esp32dev\partitions.bin
Compiling .pio\build\esp32dev\lib7e8\AsyncTCP\AsyncTCP.cpp.o
In file included from c:\users\grey\.platformio\packages\toolchain-xtensa-esp32\xtensa-esp32-elf\include\c++\8.4.0\xtensa-esp32-elf\no-rtti\bits\gthr.h:151,
                 from c:\users\grey\.platformio\packages\toolchain-xtensa-esp32\xtensa-esp32-elf\include\c++\8.4.0\ext\atomicity.h:35,
                 from c:\users\grey\.platformio\packages\toolchain-xtensa-esp32\xtensa-esp32-elf\include\c++\8.4.0\bits\basic_string.h:39,
                 from c:\users\grey\.platformio\packages\toolchain-xtensa-esp32\xtensa-esp32-elf\include\c++\8.4.0\string:52,
                 from c:\users\grey\.platformio\packages\toolchain-xtensa-esp32\xtensa-esp32-elf\include\c++\8.4.0\stdexcept:39,
                 from c:\users\grey\.platformio\packages\toolchain-xtensa-esp32\xtensa-esp32-elf\include\c++\8.4.0\array:39,
                 from c:\users\grey\.platformio\packages\toolchain-xtensa-esp32\xtensa-esp32-elf\include\c++\8.4.0\tuple:39,
                 from c:\users\grey\.platformio\packages\toolchain-xtensa-esp32\xtensa-esp32-elf\include\c++\8.4.0\functional:54,
                 from C:\Users\Grey\.platformio\packages\framework-arduinoespressif32\libraries\WiFi\src/WiFiGeneric.h:28,
                 from C:\Users\Grey\.platformio\packages\framework-arduinoespressif32\libraries\WiFi\src/WiFiSTA.h:28,
                 from C:\Users\Grey\.platformio\packages\framework-arduinoespressif32\libraries\WiFi\src/WiFi.h:32,
                 from src\simple-webserver.cpp:2:
c:\users\grey\.platformio\packages\toolchain-xtensa-esp32\xtensa-esp32-elf\include\c++\8.4.0\xtensa-esp32-elf\no-rtti\bits\gthr-default.h: In function 'int __gthread_mutex_timedlock(__gthread_mutex_t*, const __gthread_time_t*)':
c:\users\grey\.platformio\packages\toolchain-xtensa-esp32\xtensa-esp32-elf\include\c++\8.4.0\xtensa-esp32-elf\no-rtti\bits\gthr-default.h:781:12: error: 'pthread_mutex_timedlock' was not declared in this scope
     return __gthrw_(pthread_mutex_timedlock) (__mutex, __abs_timeout);
            ^~~~~~~~
c:\users\grey\.platformio\packages\toolchain-xtensa-esp32\xtensa-esp32-elf\include\c++\8.4.0\xtensa-esp32-elf\no-rtti\bits\gthr-default.h:781:12: note: suggested alternative: 'pthread_mutex_trylock'
In file included from c:\users\grey\.platformio\packages\toolchain-xtensa-esp32\xtensa-esp32-elf\include\c++\8.4.0\xtensa-esp32-elf\no-rtti\bits\gthr.h:151,
                 from c:\users\grey\.platformio\packages\toolchain-xtensa-esp32\xtensa-esp32-elf\include\c++\8.4.0\ext\atomicity.h:35,
                 from c:\users\grey\.platformio\packages\toolchain-xtensa-esp32\xtensa-esp32-elf\include\c++\8.4.0\bits\basic_string.h:39,
                 from c:\users\grey\.platformio\packages\toolchain-xtensa-esp32\xtensa-esp32-elf\include\c++\8.4.0\string:52,
                 from c:\users\grey\.platformio\packages\toolchain-xtensa-esp32\xtensa-esp32-elf\include\c++\8.4.0\stdexcept:39,
                 from c:\users\grey\.platformio\packages\toolchain-xtensa-esp32\xtensa-esp32-elf\include\c++\8.4.0\array:39,
                 from c:\users\grey\.platformio\packages\toolchain-xtensa-esp32\xtensa-esp32-elf\include\c++\8.4.0\tuple:39,
                 from c:\users\grey\.platformio\packages\toolchain-xtensa-esp32\xtensa-esp32-elf\include\c++\8.4.0\functional:54,
                 from .pio\libdeps\esp32dev\AsyncTCP\src\AsyncTCP.h:28,
                 from .pio\libdeps\esp32dev\AsyncTCP\src\AsyncTCP.cpp:24:
c:\users\grey\.platformio\packages\toolchain-xtensa-esp32\xtensa-esp32-elf\include\c++\8.4.0\xtensa-esp32-elf\no-rtti\bits\gthr-default.h: In function 'int __gthread_mutex_timedlock(__gthread_mutex_t*, const __gthread_time_t*)':
c:\users\grey\.platformio\packages\toolchain-xtensa-esp32\xtensa-esp32-elf\include\c++\8.4.0\xtensa-esp32-elf\no-rtti\bits\gthr-default.h:781:12: error: 'pthread_mutex_timedlock' was not declared in this scope
     return __gthrw_(pthread_mutex_timedlock) (__mutex, __abs_timeout);
            ^~~~~~~~
c:\users\grey\.platformio\packages\toolchain-xtensa-esp32\xtensa-esp32-elf\include\c++\8.4.0\xtensa-esp32-elf\no-rtti\bits\gthr-default.h:781:12: note: suggested alternative: 'pthread_mutex_trylock'
*** [.pio\build\esp32dev\lib7e8\AsyncTCP\AsyncTCP.cpp.o] Error 1
*** [.pio\build\esp32dev\src\simple-webserver.cpp.o] Error 1

Grey-Lancaster, 2022-06-06:
LOL Error when compiling

lyxer123, 2022-06-10:
are you solve this problem?
I think the esp32-arduio libary maybe not incompatible this source code 
down date the version？

Grey-Lancaster, 2022-06-10:
yes add @2.1.0
To use older releases, you need to specify the version in platformio.ini:

platform = espressif32@2.1.0

pidpawel, 2023-03-21:
I'm assuming that you've found the solution so I'm closing this issue. Feel free to reopen if that's not the case.

closed 2023-03-21 by pidpawel