husarnet/husarnet-esp32-platformio #4: When will husarnet be available as a library? [ QUESTION ]

zekageri, 2022-09-15:
First of all, this is an amazing piece of art.
And i think i can speak for a lot of folks here.

For now, we have to use an old release of the framework.
Obviously this is not too handy.

When will we expect to see this as a library rather then a framework?

pidpawel, 2023-03-21:
Hi! Thank you and sorry for the delayed answer.
We've already started working on making Husarnet into a library. First step was to submit pull requests to all other repositories so there's no longer need for maintaining forks. Now, after those were accepted, we've started… basically porting Husarnet to ESP32 again because of the sheer number of changes that were made both to external repositories and ours. I really hope that we can finalize this some time in the next half of 2023.
--
Have a wonderful day
Paweł

zekageri, 2023-03-22:
Thanks for the response. Good luck!

zekageri, 2024-02-12:
> Hi! Thank you and sorry for the delayed answer. We've already started working on making Husarnet into a library. First step was to submit pull requests to all other repositories so there's no longer [...]

Hello @pidpawel!

Is there any news on this issue so far?

pidpawel, 2024-02-12:
Hi!
Current status is: we have a working POC, public alpha should be available in a month or two ;)

pidpawel, 2024-02-12:
All important messages are being posted on our Community Forum (as pinned messages) (https://community.husarnet.com/). I'll try to remember to write a comment in this issue too ;)